OpenRoberta/openroberta-lab #1: Python isPrime does not work

boonto, 2019-04-17:
**Describe the bug**
 - The isPrime method in BlocklyMethods.py does not work, it tries to iterate in the range `2:sqrt(x)` and `sqrt(x)` may not be an integer.
- Tested on ev3dev

**To Reproduce**
Steps to reproduce the behavior:
1. Add isPrime block with e.g. 12
2. Run program
3. Observe float error

**Expected behavior**
 - Working program

We (Jannis and me) fixed it on ev3dev with following implementation:
```
    @staticmethod
    def isPrime(n):
        if n == 1:
            return False
        factors = 0
        for i in range(1, n+1):
            if n % i == 0:
                factors += 1
        if factors == 2:
            return True
        return False
```

boonto, 2019-07-26:
Is it added in ev3dev?

Microbit also still has "implementation missing" for isPrime, because we do not have BlocklyMethods there. But maybe that's another issue.
We could either:
 - show an "unsupported" message
 - remove it
 - add a helper method when the block is used using the hardware collector

bjost2s, 2019-07-27:
This solution is not sufficient and should be improved with the next release (2.5.1). We should also take this as an example for the redundancy of blocklyMethods and create a common issue for this where we at least figure out

- which plugins use blocklyMethods and for what
- identify plugins that do not use blocklyMethods, but have need for it (e.g. microbit)
- check if and how we can get rid of them!

In edison we add needed methods at the end of the generated code as functions, which seems to be a good and also common solution.

boonto, 2019-08-20:
Due to the inclusion of isPrime to the helper methods of #95, this issue is now a duplicate.

closed 2019-08-20 by boonto