OpenRoberta/openroberta-lab #6: USBProgram: Improve adding additional Arduinos UX

boonto, 2019-04-18:
Currently, users have to manually enter console commands and edit the configuration file to add additional vendor and product IDs to the USBProgram.
This should be changed to be more accessible.

### Concept
1. After clicking the small robot in the corner or 30 seconds without a found robot an information popup appears. Possibly in the style of a speech balloon. (thanks @VinArt)
2. The popup asks about connection problems and the type of robot that is used.
3. The users selects the respective robot:
  - EV3: the user is sent to the appropiate Wiki entry with additional information. (Localization!)
  - Arduino: A new window appears containing:
    - A text field with the output of a tty/COM port search
    - Instructions to read the output (Name may not contain Arduino), connect->reconnect to find out what changes
    - The contents of the arduino-ids.txt file
    - A button to add the selected device to the list, together with a dropdown with the Arduino type
    - (A button to remove entries)

The list could also be updated via a call to the server.

This does not affect how users currently interact with the USBProgram.

philippmaurer, 2019-05-27:
Found a bug on it:

**Problem**
If you use the device id editor to rewrite an exisiting id combination for a device or add a new device, the new values are neither saved to the "arduino-ids.txt" file, nor are they respected in the Open Roberta USB program.

**Steps for reproduction**
1. Start the Open Roberta USB program on a Windows (10) machine.
2. Open the device id editor
3. Change any values
4. Save and exit the device id editor.
5.a.1. Restart the Open Roberta USB program
5.a.2. Open the editor again and see that all changes are lost.
5.b.1. Connect a robot for which you just changed the set system to your computer.
5.b.2. See, that whatever you entered in te device editor is ignored (even without a restart) and that the data specified in the file are used instead.

**Possible solution**
Make sure that the editor saves the changes to the file.

**Environment**
Tested with an mBot and a Windwos 10 PC

boonto, 2019-05-31:
I think the id file has to be saved separately in the user directory. Right now it is saved in the program folder and I guess requires admin rights to edit. Thanks for letting me know, I will fix that ASAP.

boonto, 2019-06-05:
The problem in your comment should be fixed now.

bjost2s, 2019-06-12:
@boonto please comment with the commit where the fix can be seen. Is it already released? Then please mark as resolved and close it

boonto, 2019-06-13:
@bjost2s the commit is https://github.com/OpenRoberta/robertalab-usbprogram/commit/9257e29739aabb15ebe13aaf260301a703767778 and it's currently released in the [beta.2](https://github.com/OpenRoberta/robertalab-usbprogram/releases/tag/v3.4.0-beta.2) of v3.4.0.
I will close it as soon as it is released in the stable version.

closed 2019-06-26 by boonto