OpenRoberta/openroberta-lab #11: Write and archive log messages in separate files

vlebedynska, 2019-04-23:
**Feature request:**
Currently all log messages are stored in one file. To evaluate the statistics, we need to retrieve the statistical information from the main log file as needed. 

To facilitate this process and for evaluation purposes in the future, statistics logs should be stored in a separate file. Info logs should also be written to a separate file.

For easy maintenance, log files should be archived automatically at the end of the month and automatically supplemented with continuous logs when the server is restarted.

**Solution:**
1. Separate statistical log messages from other log messages.
2. Save logs in separate files (e.g. using `RollingFileAppender`)
3. Roll log files based on date/time.
4. Daily rollover (at midnight) for testing purposes.

rbudde, 2019-05-21:
I added two properties to the server properties, that allow to inject at runtime the log level and the logback configuration file (have a look at the comments in openroberta.properties)

rbudde, 2019-05-27:
tested by inspecting the log directory, if logback-prod.xml is used for logging

closed 2019-05-27 by rbudde