OpenRoberta/openroberta-lab #12: bug in RGB function (Arduino RobertaFunctions)

springob, 2019-04-24:
**Describe the bug**
 -RGB-Eye-color is buggy...

**To Reproduce**
Steps to reproduce the behavior:
Test 'dark green' (3rd color in drop-down) with BOB3

**Solution**
replace code in OpenRobertaParent/RobotArdu/resources/libraries/RobertaFunctions/NEPODefs.h

wrong code:
inline unsigned RGB(unsigned r, unsigned g, unsigned b) {
    return (((r & 0xF8) << 8) | ((g & 0xFC) << 3) | ((b & 0xF8) >> 3));
}

correct code:
inline unsigned RGB(unsigned r, unsigned g, unsigned b) {
  return ((r&0xF0)<<4) | (g&0xF0) | ((b&0xF0)>>4);
}

springob, 2019-04-24:
BOB3 uses 12 bit RGB Color (4+4+4), the function calculates a 16 bit value (5+6+5)...

bjost2s, 2019-04-25:
Thanks for this issue.
NEPODefs.h is used in all Arduino plugins and we already have discovered another issue with this function for e.g. Uno:
[image]
Because of reducing the color information the assertion in the example above will **not** fail.

closed 2019-06-27 by VinArt