OpenRoberta/openroberta-lab #21: Caliope Simulation: assignment of number to Variable with value null not performed

kwollw, 2019-05-12:
given the following NEPO program for Calliope:
[image]

At the calliope board "Zeige Text x" displays "1" but the simulation displays "undefined".

Without  initialialization of "null" the simulation works correctly.
[image]

It should be expected that board and simulation show identical behavior.

bjost2s, 2019-08-05:
see #74

rbudde, 2019-08-07:
see #215

bjost2s, 2019-09-23:
still an issue, not solved in #215

bjost2s, 2019-10-02:
proposed solution: null may only be used for reference types. Constrains should be checked in blockly and visitor

simonedegiacomi, 2019-10-02:
> proposed solution: null may only be used for reference types. Constrains should be checked in blockly and visitor

I think that this is a good idea, since it would also fix another bug: If you try to assign 'null' to a number variable using lejos, you get a compilation error:
[image]
[image]

If you want I can work on this issues. The only question I have is if the solution should apply to blockly and visitor of only Caliope (and maybe lejos) or to all the robots.