OpenRoberta/openroberta-lab #25: Support for Pixy Cam for EV3 (maybe NXT)

bjost2s, 2019-05-17:
**Is your feature request related to a problem? Please describe.**
 - A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
 - A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

ensonic, 2019-05-31:
@bjost2s please actually fill the above template :)

bjost2s, 2019-07-18:
** + Integrate Pixy-classes into c4ev3,** maybe later

bjost2s, 2019-11-13:
next TODO: check the branches from @RiErdmann, if we easily can integrate this feature

closed 2024-11-05 by bjost2s