OpenRoberta/openroberta-lab #31: c4ev3 text to speech

simonedegiacomi, 2019-05-19:
Parent issue: #27 

- [ ] add espeak and sox compiled for EV3 to ora-cc-rsc
- [ ] update UI to download easpeak and sox binaries to sd card/usb drive
- [ ] add to c4ev3 a function that:
	1. uses espeak to generate wav file (RIFF (little-endian) PCM, 16 bit, mono 22050 Hz)
	2. converts generated wav to compatible wav (RIFF (little-endian) PCM, 8 bit, mono 8000 Hz)
	3. plays wav file

simonedegiacomi, 2019-06-20:
>     * [x]  add to c4ev3 a function that:
>       
>       1. uses espeak to generate wav file (RIFF (little-endian) PCM, 16 bit, mono 22050 Hz)
>       2. converts generated wav to compatible wav (RIFF (little-endian) PCM, 8 bit, mono 8000 Hz)
>       3. plays wav file
This method is very slow, try to
- [ ] generate/find espeak language files already encoded in (RIFF (little-endian) PCM, 8 bit, mono 8000 Hz) to avoid the need of sox

ensonic, 2019-06-21:
FYI: there is https://github.com/espeak-ng/espeak-ng/issues/88 for espeak to support other sample rates.

simonedegiacomi, 2019-06-24:
thank you! I didn't see the issue. If I understood it correctly the feature isn't implemented yet, is it?
I was thinking about using [espeakedit](http://espeak.sourceforge.net/editor.html) to recompile the espeak data directly to the needed sample rate, what do you think? doing this we would avoid the cost of converting the audio files at runtime

closed 2019-07-30 by boonto