OpenRoberta/openroberta-lab #35: Calliope: name clashes (e.g. "sum")

rbudde, 2019-05-21:
Using "sum" as variable name gives a redeclaration error (with the function "sum" from MicroBit.h:41) and the program cannot be compiled

**Expected behavior**
- this name clash and all other clashes of this kind should be avoided.
- if not possible, at least the compiler message has to be shown to the user (maybe he/she can then avoid the problem)

**General remark**
 - a clear concept of separating user-defined names and ora-defined names from "standard library" names is needed

VinArt, 2019-07-01:
this is resolved for calliope now by adding 3 underscores in front of user variables. If this looks fine and is accepted, I will extract it higher up for all the robots.

VinArt, 2019-07-02:
resolved for all plugins.

boonto, 2019-07-24:
Do we need this for non C based robots as well?
Reserved words in microbit are suffixed with 2 automatically, however, they are not suffixed for EV3dev, e.g. variables can be named `def`.
For Java you can name variables "List" which also leads to problems.

bjost2s, 2020-01-03:
This fix has many commits, so very difficult to check all of them, but seems to work.

closed 2020-01-03 by bjost2s