OpenRoberta/openroberta-lab #38: Remove all cookies from the lab so that the cookie disclaimer could be removed too

bjost2s, 2019-05-21:
**Is your feature request related to a problem? Please describe.**
 - I'm always frustrated when I open the Lab, create a program and want to start compiling the program e.g. for Calliope, because I think I have to remove the cookie disclaimer to have a full view. Also I do not want to accept cookies in general. 

**Describe the solution you'd like**
 - Remove all cookies and the cookie disclaimer from the website. To remember the choice of the robot type we should offer a simple creation of a corresponding bookmark.
**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.
We discussed this often, but because of some legal stuff this seems to be the best option. Also we recently got aware of that there is no need to use cookies for Matomo in general.

**Outlook**
 - With this feature we should start thinking about collecting statistical data (**not personal data**) and provide them to everyone via the UI, somehow similar to what Matomo is doing now for us.

boonto, 2019-12-20:
Steps to do:
- [x] Remove cookies
- [x] Delete previous cookies
- [ ] Show popup to create a bookmark
- [ ] Fitting message depending on the browser/device
- [ ] Adapt loadSystem call to standard query structure
- [ ] (Later on) remove jquery-cookie library

boonto, 2019-12-27:
I decided to use a generic message instead of generating one depending on the device/browser.

bjost2s, 2020-01-02:
reviewed but not closed, because of open items to do later, therefore removed from the current project (to be added to the following one)

boonto, 2020-01-17:
Due to our usage of local storage for background images we adapted it a little bit:

Scenario cookies disabled:
- User uploads an image
- Image is uploaded to DOM
- Image is cleared after the session

Scenario cookies enabled:
- User presses upload button
- Popup with local storage info show up
- On confirm the image is stored across sessions
- On deny the image is cleared after the session as above
- Local storage images older than 30 days are automatically removed
- When cycling to the stored background image the timer is reset

bjost2s, 2020-01-28:
rest work will be continued in #710

closed 2020-01-28 by bjost2s