OpenRoberta/openroberta-lab #41: Calliope Motor-Block with Simulator

Thorlogy, 2019-05-24:
**Describe the bug**
 - The Motor-Block does not work with the simulator

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 - A clear and concise description of what you expected to happen.

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2019-06-12:
This block is not supported by the simulation because it is part of the "differential drive". We will show a hint to this block like we do it for other blocks, when the simulation is executed.

boonto, 2019-09-30:
I cannot reproduce this with the depicted block in 3.6.0. When using it, both "motors" in the simulation start turning.
Only the "Calli:bot" modes do nothing in the simulation. Should the "not supported" message be displayed when using these modes?

The ones that do not work in the simulation are these:
[image]
But they have the correct "not supported" message.