OpenRoberta/openroberta-lab #45: Can not open simulation, if the "to (var) append test (string)" block is used

philippmaurer, 2019-05-28:
**Describe the bug**
It is not possible to open or start the simulation, if the "to (var) append test (string)" block is used in the program. The simulation either wont open, or it will not start, if it was opened before the block was part of the program.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Choose one of the following robot systems: NXT, EV3, micro:bit, calliope (any with a simulation)
2. Open or create a program, that uses the "to (var) append test (string)" block. You can fill the parameters of the block with any legal or illegal combination of strings and variables.
3. Open the simulation. It will not work.

**Expected behavior**
The block should work as intended in the simulation. It does work on the robot, though.

**Screenshot**
[image]

**Device information**
- Environment: Test and production
- Tested if the block works on the robot: NXT (works)

rbudde, 2019-06-25:
Works now on both Calliope and simulator. I consider the code generation for the Calliope as a temporara solution, because NO check is done, whether the left hand parameter is a variable (must be). If it is a literal, the following problematic code is generated:

`ManagedString("a") = ManagedString("a") + ManagedString(ManagedString("a"));`

The check visitor should detect this and emit an error

rbudde, 2019-06-25:
Issue should be moved to the next relaese project

closed 2019-07-01 by VinArt