OpenRoberta/openroberta-lab #46: Refactoring ConfigurationComponent to have subclasses for Actor/Sensor/Asset

varunshankar, 2019-05-28:
Refactoring configuration component to have subclasses for Actor/Sensor/Asset

ensonic, 2019-06-24:
According to gitter discussion, the way forward will be:
1.) Create one pull request that does the refactoring. This can be two changes - one for extracting Actor/Sensor subclasses from ConfigurationComponent and making ConfigurationComponent a baseclass. The 2nd change on this PR will make the rewrites of all the constructor calls. Ideally this is does with a script (eg "git grep" + "sed"). This is to make sure if the that the 2nd part can be redone as needed.

2.) If the previous PR is merged, a new subclass for ConfigurationAssetComponent can be added

rbudde, 2023-06-05:
a large commit. Parts have been reviewed.

closed 2023-06-05 by rbudde