OpenRoberta/openroberta-lab #47: TypecheckVisitor.java needs a useful javadoc comment

ensonic, 2019-05-31:
The javadoc comment here
https://github.com/OpenRoberta/openroberta-lab/blob/master/OpenRobertaRobot/src/main/java/de/fhg/iais/roberta/typecheck/TypecheckVisitor.java#L113

is not useful. I propose the replace
```
This class is implementing {@link IVisitor}. All methods are implemented and they append a human-readable JAVA code representation of a phrase to a
StringBuilder. <b>This representation is correct JAVA code.</b> <br>
```
with
```
A helper class for unit tests to validate that all ASTs built from blockly XML are using compatible types.
 *
 * <p>This helper can be used in tests to validate the constraints on blockly toolboxes (generic ones and robot specific
 * ones).</p>
```

and also fix some other javadoc issues and typos in this class.

VinArt, 2019-05-31:
I think, that this can be used not only in tests, but in general, for verification before code generation, so yes, the java doc should be altered, but maybe not mention that this is only test-related?

ensonic, 2019-05-31:
Right now we are only using it in tests. I think the only way that this could find errors is that we have bad blocks and hence my suggestion that this is mostly useful for tests.