OpenRoberta/openroberta-lab #55: As a user I'd like to leave a group

elhele, 2019-06-03:
Precondition: the user is logged in

People: user, group admin

Subjects: group

1. From the menu "user" user chooses "go to groups".
2. User is redirected to a groups tab with the list.
3. User can leave one, several or all the groups:
- to leave one group user presses a leave button in the line of this group in the table
- to leave several groups user ticks the groups to be deleted and presses leave button on the top of the table
- to leave all the groups those belong to the user, user ticks a checkbox on the top of the table and presses delete on the top of the table.
4. The system responses with a popup that contains the following:
- question if the user really wants to leave the group(s)
- name of the group / list of the names of the groups
- button to confirm and button to abort
5. The user confirms the deletion by pressing the corresponding button.
6. The system checks if everything is correct and returns:
- a message that he/she left the group if everything is fine; user will also immediately see that the group disappeared from the group list
- in case the user was an admin, he/she will see a popup with a list of the group members, from whom he/she chooses the new admin, and confirmation and abort buttons.
- a popup with the error description in case if group deletion cannot be performed. This case is unlikely and can happen only because of some database or server error.
7. The user loses access to shared programs and the programs shared by users are not shared with the group anymore.
8. Admin of the group receives a popup that the user left the group.

This can also be used as a test case.

This issue is related to #384

bjost2s, 2019-06-12:
I think we do not want to offer this feature, please check with https://github.com/OpenRoberta/openroberta-lab/wiki/Group-Accounts

bjost2s, 2019-06-30:
@vlebedynska, I close this issue for you

closed 2019-06-30 by bjost2s