OpenRoberta/openroberta-lab #67: Wedo: Can't select note on ipad

Dittse, 2019-06-12:
**Describe the bug**
 - you can't select another not than c1 on Ipad

**To Reproduce**
Steps to reproduce the behavior:
1. Open the open roberta mobile app
2. get the play note block
3. try to select another note
4. you can't.

**Expected behavior**
 - You can select another note

**Screenshots**
 - If applicable, add screenshots to help explain your problem.
[image]

**Device information**
 - Thorstens IPad 4
 - Open Roberta Mobile App Version 1.0.0

philippmaurer, 2019-07-16:
I just reproduced this bug on the Roberta Mobile App and mobile Safari.
iPad 4th. Gen.
iOS: 10.3.3 (14G60)

bjost2s, 2024-10-25:
solved for mobile safari, web view in open roberta app can't be changed anymore.

closed 2024-10-25 by bjost2s