OpenRoberta/openroberta-lab #71: serial write/read for Calliope, microbit and arduino

rbudde, 2019-06-12:
we need serial write/read for Calliope, microbit and arduino, see #1007

Still missing:
- [x] Write serial implementation in Calliope visitor
- [x] Write serial implementation in Microbit visitor
- [x] Write serial implementation in arduino visitor
- [ ] Write serial implementation in simulation visitor
- [ ] Adding read serial block to all toolboxes
- [ ] Read serial implementation in Calliope visitor 
- [ ] Read serial implementation in Microbit visitor 
- [ ] Read serial implementation in arduino visitor 
- [ ] Read serial implementation in simulation visitor 
- [ ] Tests

bjost2s, 2020-04-23:
Reopen because several users are asking for serial read(). 

This would add a convenient option to interact with the devices.

bjost2s, 2024-10-24:
no time to work on it

closed 2024-10-24 by bjost2s