OpenRoberta/openroberta-lab #74: Rewrite simulation code generation and execution in the WeDo style

bjost2s, 2019-06-12:
**Is your feature request related to a problem? Please describe.**
 - We have a couple of open issue in simulation execution: #1116, #1053, #1008, #1153, #1003, #1002, #981, #903
**Describe the solution you'd like**
 - Rewrite the code generation and execution like we did it already for WeDo (not used for simulation but for steering/triggering WeDo via bluetooth). With this stack based way we expect to get rid of a couple of issues that we have with the old simulation. Also some issues we cannot fix with the old way.

**Additional context**
 - see and get more information from the [WeDo intepreter project](https://github.com/OpenRoberta/robertalab/tree/master/OpenRobertaParent/WedoInterpreter)

rbudde, 2019-08-06:
see also #215 for a collection of errors

rbudde, 2019-08-07:
all errors mentioned here are now linked in #215. This ticket is closed.

closed 2019-08-07 by rbudde