OpenRoberta/openroberta-lab #82: Add servo / PWM block to calliope

jischebeck, 2019-06-12:
Currently using Roberta Labs it is not possible to configure an output as PWM to control a servo.
Add a block to control a servo to the calliope mini.

"set servo at <P1> to Hz <1000>"

The functionality already exist in Microsoft MakeCode.

rbudde, 2020-02-10:
checked, but the following remarks:
- i don't understand the text `set servo at to Hz <1000>` from the description
- i don't understand the first comment

I believe, that the servo block here is using PWM to set a servo motor to a well-defined degree.

I don't close this issue, because:
- the english tooltip is completely wrong
- the help block (on the right side) is missing

B.T.W: this will be deployed today IN THIS BAD state.

rbudde, 2020-02-11:
The block is designed to

use PWM to set a servo motor to a well-defined degree

the english tooptip is correct now

closed 2020-02-11 by rbudde