OpenRoberta/openroberta-lab #83: Pasted blocks inherit the 'inTask' property of the copied block

VinArt, 2019-06-12:
**Describe the bug**
 - If an already attached block is copied and pasted, it stays active despite the fact that it is not connected anymore

**To Reproduce**
Steps to reproduce the behavior:
1. Copy-paste attached block
2. Block is active (not greyed out)
3. If at this point code is regenerated it is generated in a wrong way

**Expected behavior**
 - block would be grey

**Screenshots**
[image]

PamoCurtis, 2022-03-11:
This Issue is similar to [that one](https://github.com/OpenRoberta/openroberta-lab/issues/1204#issue-1110506038) and also should be fixed with this [pull request](https://github.com/OpenRoberta/blockly/pull/61#issue-1166201746)

bjost2s, 2022-05-24:
fixed with #1204

closed 2022-05-24 by bjost2s