OpenRoberta/openroberta-lab #95: Replace usages of BlocklyMethods

boonto, 2019-06-12:
In Java and Python code generation we use `BlocklyMethods` for list and math operations.
In the course of #1011 the list operations in `BlocklyMethods` will be replaced by inline code or readable methods in the program.
As there is (currently?) no discernible reason why we should use the non-transparent `BlocklyMethods`, e.g. `BlocklyMethods.sin(x)` instead of simply `Math.sin(x)`, all of these calls should be replaced.
This will also increase readability and accessibility of the generated code.

ensonic, 2019-06-23:
For ev3dev this is the impl: https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/BlocklyMethods.py
Lots of them are one-liners. Inlining them makes a lot of sense. For the more complex ones we could emit the helper function as needed and call it. Getting rid of the class in the long run makes a lot of sense to me.

boonto, 2019-08-08:
Python observations:
Microbit:
 - does not use BlocklyMethods.py (BM)
 - already implements many of the concepts we want: mosty one liners, attached methods (stdDev, median)
 - GOLDEN_RATIO is a constant that is always defined
 - isPrime implementation is missing
 - randomOfList is missing
 - all of the one liner implementations could be pulled up

EV3dev:
  - GOLDEN_RATIO uses BM
  - isEven/isOdd/.. uses BM
  - math on lists uses BM
  - clamp uses BM
  - random methods use BM
  - random methods are implemented using os.urandom with comment that random module is too large -> same methods should be used across robots

NAO:
  - identical to EV3dev
  - color uses BM

Vorwerk/Raspberry:
  - identical to EV3dev

For python almost everything can be pulled up to the generic code gen visitor.

uzkns, 2019-08-09:
For Python:
[blockly_methods_all_robots.zip](https://github.com/OpenRoberta/openroberta-lab/files/3485232/blockly_methods_all_robots.zip)
For C:
[NEPO_all_robots.zip](https://github.com/OpenRoberta/openroberta-lab/files/3485238/NEPO_all_robots.zip)

boonto, 2019-08-20:
Implemented for Python, the helper methods are saved in a YAML file for now. If necessary, it can be split into individual language files later (Python only needs 3 helper methods).

Individual robots may override higher up implementations of methods.

boonto, 2019-08-23:
Will be continued in #230.

VinArt, 2019-08-30:
- _isPrime is not generated when block used inside the loop
- _median is not generated when block used inside the loop
- _standard_deviation is not generated when block used inside the loop

tested on microbit and ev3dev

boonto, 2019-08-30:
Ah thanks, 
```
repeatStmt.getList().visit(this);
```
is missing [here](https://github.com/OpenRoberta/openroberta-lab/blob/develop/OpenRobertaRobot/src/main/java/de/fhg/iais/roberta/visitor/collect/ICollectorVisitor.java#L248).

bjost2s, 2019-09-02:
@VinArt please fix this

closed 2019-09-04 by VinArt