OpenRoberta/openroberta-lab #101: Calliope Nachrichten verschicken

Thorlogy, 2019-06-18:
**Describe the bug**
 - Das Programm "hängt sich auf" nachdem der A Knopf gedrückt wurde. Wird der B Knopf gedrückt passiert nichts (LED geht nicht an). Das selbe Programm ohne den Nachrichten Block funktioniert wie erwartet. Werden die Nachrichtenzahlen in Variablen geschrieben und dann genutzt funktioniert das Programm.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 - Das Programm zeigt je nachdem welcher Knopf gedrückt wurde, die entsprechende LED Farbe. Und schickt eine Nachricht.

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

[image]

**Device information**
 - Laptop Chrome - Chrome Browser
 - Calliope 2018

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2019-06-19:
The program shown on the screenshot only reacts on button A.

bjost2s, 2019-06-24:
Do you need more help or should we or you close this issue?

Thorlogy, 2019-06-25:
Nun mit korrektem Screenshot - bug siehe wie beschrieben oben. 
[image]

bjost2s, 2019-06-26:
Still cannot be reproduced on my calliope 2017. Working as expected. Could you please add the exported   xml file and maybe a video?