OpenRoberta/openroberta-lab #102: Calli:Bot Auswahl mit Port B

Thorlogy, 2019-06-19:
**Describe the bug**
 - Es scheint nicht möglich zu sein, im Motor-Block Port B auf calliBot zu stellen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'or lab'
2. Click on Calliope
3. Scroll down to 'Aktion, Motor Block
4. See error, when try to select calliBot rechts at Port B

**Expected behavior**
 - CalliBot rechts lässt sich auswählen

**Screenshots**
 
[image]

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2019-06-19:
No, it is not possible, because Calli:Bot has a fixed base without port A and B. Calli:Bot has a right and left motor with fixed ports as available in the block. Please check the documentation once it is available.
Or have a look on the configuration of Calliope, there you can see where Port A and B are located.
[image]

VinArt, 2019-06-20:
I think the issue here is in that there should be symmetry in the way one chooses the callibot motors, so it would not matter from which port the change is made, in any case both should be changed to callibot motors.

Thorlogy, 2019-10-13:
Any progress?

boonto, 2020-01-10:
Works now.

closed 2020-01-10 by boonto