OpenRoberta/openroberta-lab #103: Programmname

Thorlogy, 2019-06-19:
**Is your feature request related to a problem? Please describe.**
 - A clear and concise description of what the problem is. Ex. I'm always frustrated when i export a progam and the name of the program is not desplayed. All exported program have the Standard Name NEPOprog

**Describe the solution you'd like**
 - Display the correct name

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

bjost2s, 2019-06-19:
This feature is already implemented, please load a program from the database, export it and the exported program has exactly the same name as the one from the lab.
[image]

ensonic, 2019-06-21:
If you use the lab as a not-logged in guest, you can't rename the program and it will be downloaded as NEPOProg.xml. If you rename the saved file and re-import it, then name is correctly shown. Logged in users can name the program when storing it in the cloud.

Thorlogy, 2019-06-25:
Nevertheless, it is confusing for the user if he/she imports a program with a name and then finds that the program has the name NEPOprog. Often this is not immediately noticeable, but only after a certain time.

bjost2s, 2019-07-02:
see #147

closed 2019-07-02 by bjost2s