OpenRoberta/openroberta-lab #109: Configuration for all robots - visible or hidden

bjost2s, 2019-06-26:
Currently we have robots with the old configuration and some robots without a configuration (maybe internally on server side they already have a fixed configuration). 
We should add to all systems the "new" configuration, so that we can create a common way on server side to handle this configurations. Further we have than the possibility to assign multiple  "standard"  configuration to e.g. calliope to distinguish between different calliope like systems. A good example is here Calli:Bot:
- standard configuration for calliope itself
- Calli:Bot configuration to configure the maqueen board, the underlying board of the Calli:Bot
- another callibot configuration for the next calliope robot (there are several out there, mainly with an external motor driver board)
- ...
The user can then choose in the beginning the "configuration" while he is choosing e.g. Calli:Bot as a system - or - if we do not want to offer to much systems the user has to make this choice in the configuration itself (might be difficult to show this opportunity to the user)

boonto, 2020-04-24:
We also discussed an additional point: The current images describing all sensors/actors/ports on the robot should be accessible through an additional tab, e.g. "robot description". This would also require generating new images

Additional tasks:
- [ ] 
- [ ] 
- [ ] 
- [ ]

boonto, 2020-07-07:
Calliope and Microbit were handled in #747.
Remaining robots with visible configuration:
- [ ] EV3
- [ ] NXT
- [ ] mBot
- [ ] Botnroll
- [ ] Vorwerk
Remaining robots with invisible configuration:
- [ ] BOB3
- [ ] Edison
- [ ] NAO

rbudde, 2022-05-11:
will be solved with new issues, if really needed

closed 2022-05-11 by rbudde