OpenRoberta/openroberta-lab #110: Tests for our library functions / header files

rbudde, 2019-06-26:
**Rationale**
we found too many errors in our own library functions in the past. This reduces usability of the lab. These errors let fail correct user program and are very confusing for users.

**Solution**
for all robots, that use native code generation, we need test programs in the target language (C/C++, Java), that test our library functinos and header files. For python, we should consider to add a lint tool (already proposed elsewhere) for checking generated python (including the imports, of course)

boonto, 2019-11-11:
I just noticed something regarding the microbit integration tests: the compiler always passes regardless of eventual python errors in the program. This makes our current integration tests for microbit useless. We should definitively add a linter that checks for basic python rules for the tests (and maybe the user).

A special case for microbit is when the uflash library is missing, then instead of putting the hex in the file the error message is inserted instead. We could check for that and fail the tests in that case, making running the integration tests a valid option to see if all robots would work on your system.

bjost2s, 2022-05-24:
fixed exept python linter

closed 2022-05-24 by bjost2s