OpenRoberta/openroberta-lab #114: visitors to check program and conf validity for all robots

rbudde, 2019-06-26:
**Reason**
- the conf and the program must match. If this is not the case, an error has to be reported
- the program must be complete (e.g. parameter of an expr block may be missing, i.e. the block shows a "hole". An error message has to be reported for an incomplete program.

**Remarks**
- a consistent solution for all robots is required
- we have old and new configurations. Refactoring of that may be needed
- comments on the steps to solve this issue are welcome

VinArt, 2019-07-17:
- [ ] sim
- [ ] wedo
- [ ] ev3lejosv1
- [ ] ev3lejosv0
- [ ] ev3dev
- [ ] nxt
- [ ] vorwerk
- [ ] microbit
- [ ] botnroll
- [ ] nao
- [ ] bob3
- [x] uno
- [x] nano
- [ ] mega
- [ ] sensebox
- [ ] mbot
- [ ] calliope2017
- [ ] calliope2016
- [ ] calliopeSim

VinArt, 2019-07-31:
- Validators are/will be implemented as visitors to the configuration or configuration components. Currently configuration validator works on the separate configuration components, though this should be changed so it would work on the whole configuration.
- Validators classes are loaded through reflection, and are checked to contain "visit" method, so to make sure that they are valid visitors/for early fail.
- Configuration can now be checked and annotated, since ConfigurationComponents now extend Phrase. Visit method in Phrase may not be needed, so it would not be inherited by sub classes. Maybe ConfigurationComponent should not extend Phrase at all, instead contain needed information for later JAXB reconstruction.
- Annotated configuration is returned to frontend in the JSON object under "configuration" field. "data" field should be renamed to "program" since it contains the blockset of program. Maybe field names should actually reflect that their contents is blockset, so to have the names like "program_blockset" and "configuration_blockset"
- Phrase has naming inconsistency with respect to visitor pattern (accept method is called visit) this has effect on many classes and should be resolved after GSoC is over.
- Frontend should deal with the configuration received from the server in a similar way to program, the given blockset should be converted to the workspace and annotation should be shown. Maybe better names for program blockly workspace and configuration blockly workspace should be chosen (instead of  bricklyWorkspace and blocklyWorkspace).
- If errors in configuration are encountered while in the program tab, configuration tab should indicate their presence by adding visual annotation (either flashing colour or icon or both)
- In the configuration tab error annotations should be closed when the user makes modifications to the configuration not to obstruct the view, though the cross symbols should stay in order for the user to be able so see exact errors. Annotations will be removed by the next configuration blockset conversion to blockly configuration workspace if errors were corrected.
- Validators should be pipelined, so they would work on the previously processed data. Validators (and other visitors) should not throw exceptions and only add annotations to the visited blocks. Most probably validators should be chained.
- Frontend should always deliver a configuration, since it is loaded anyway and not provided the name exclusively. The server then would be able to retrieve all the needed data for source code generation or program compilation from the request without the need to fetch the configuration from the database again. This would simplify the REST endpoint.
- When the visitors are written in the way they can be pipelined, then workflows should be defined in a way that needed visitors are selected and executed, leaving the REST endpoint with only a single call to the workflow that would accept data from the request and provide data for the response. This data is dependent on the workflow and may include source of the program, program and configuration block sets or URI(s) to artefacts (program binaries)

VinArt, 2019-08-02:
- Validator visitors will now have a "validate" method instead of "visit"
- Validators will work on a Project, which combines Program and Configuration. This is currently known as BlocklyProgramAndConfigurationTransformer, but will be extended with the results of validation and resulting keys.
- An entry is intriduced in robot properties for a comma-separated list of visitors/validators for a pipeline. Currently only one is used, but in general for each workflow there must be a pipeline of visitors defined.
- Validators should be designed to be stateless and operate on the shared Project that would contain the results of validation.
- Project instances will be created for each request and will thus refresh data needed for the Validators to work properly, for example freePins for Arduino will be re-set.
- Validator instantiation is moved from CompilerWorkflow to AbstractRobotFactory. When errors are encountered during instantiation of Validator classes the server will not start producing an exceptional situation.
- REST interface should be redesigned in a way that it has small controller classes which deconstruct the request and fish out all needed parameters and then execute the corresponding workflow. For example a executeSourceCodeWorkflow is called for /project/source endpoint and produces source code+ annotated project. Further workflows are yet to be defined with regard to existing switch-case on cmd statements. For this to work properly additional entries in property file must be made to define correct validation/visiting pipelines.
- In general we will have 2 stages of project processing: validation and compilation.
- v1 and v2 packages are created to separate different API versions, though the actual version will be most probably dine via the request headers.

VinArt, 2019-09-09:
- validators are now known as workers, they provide a general wrapper-concept for visitors, that implement the desired behaviour
- workflows consists of a sequence of workers, that work on a Project
- workflows are executed via one static method, that accepts the workflow name
- show source workflow implemented
- run workflow implemented unifying runP and runPBack commands

rbudde, 2019-11-12:
was tested. Remaining work put into new issue #306

closed 2019-11-12 by rbudde