OpenRoberta/openroberta-lab #123: Login -> Change -> Logout -> New account leads to server error

boonto, 2019-06-26:
After this specific sequence of events the server returns a "ORA_USER_UPDATE_ERROR_ACCOUNT_WRONG" error:

1. Login as a user
2. Change account information
3. Close the change screen
4. Logout
5. Create a new account

Everything has to happen in the same session.

rbudde, 2020-02-04:
looks like a frontend problem, that could hit other services as well. User change command crashes in the frontend. Probably the state is destructed. Following new account is considered an update (without an user logged in). The server message is ok, as the account is '', which is not allowed. Logging ist:

```
2020-02-04 09:30:22,841 [qtp1464191502-41] session-id=[1] user-id=[67795] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 3 msec: update user 'qay1' to server
2020-02-04 09:30:22,841 [qtp1464191502-41] session-id=[1] user-id=[67795] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[ERR ]] [[TIME]] 0 msec: got user info from server, then EXCEPTION: TypeError: successFn is not a function
2020-02-04 09:30:50,989 [qtp1464191502-43] session-id=[1] user-id=[67795] robot-name=[calliope2017NoBlue] INFO d.f.i.r.j.r.a.controller.ClientUser - command is: logout
2020-02-04 09:30:50,989 [qtp1464191502-43] session-id=[1] user-id=[67795] robot-name=[calliope2017NoBlue] INFO d.f.i.r.j.r.a.controller.ClientUser - logout of user 67795
STATISTICS {"time":"2020-02-04 09:30:50,989", "sessionId":"1", "robotName":"calliope2017NoBlue", "logLevel":"INFO", "message":{"action":"UserLogout","args":{"success":true}}}
2020-02-04 09:31:17,932 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 2 msec: user clicked
2020-02-04 09:31:17,932 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"logout","rc":"ok","message":"ORA_USER_LOGOUT_SUCCESS","serverTime":"Tue Feb 04 09:30:50 CET 2020","server.version":"3.8.2-SNAPSHOT","initToken":"u_c6_1U_ZSa4"}
2020-02-04 09:31:17,932 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 13 msec: logout user
2020-02-04 09:31:17,932 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 4 msec: user clicked
2020-02-04 09:31:17,932 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.j.r.a.controller.ClientUser - command is: updateUser
2020-02-04 09:31:17,934 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.p.AbstractProcessor - Processor succeeded: USER_GET_ONE_SUCCESS
2020-02-04 09:31:17,934 [qtp1464191502-41] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.p.AbstractProcessor - Processor failed: USER_UPDATE_ERROR_ACCOUNT_WRONG
2020-02-04 09:31:42,836 [qtp1464191502-43] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 2 msec: update user '' to server
2020-02-04 09:31:42,837 [qtp1464191502-43] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.j.r.a.controller.ClientUser - command is: updateUser
2020-02-04 09:31:42,839 [qtp1464191502-43] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.p.AbstractProcessor - Processor succeeded: USER_GET_ONE_SUCCESS
2020-02-04 09:31:42,839 [qtp1464191502-43] session-id=[1] user-id=[-1] robot-name=[calliope2017NoBlue] INFO d.f.i.r.p.AbstractProcessor - Processor failed: USER_UPDATE_ERROR_ACCOUNT_WRONG
```

rbudde, 2020-02-10:
checked

closed 2020-02-10 by rbudde