OpenRoberta/openroberta-lab #124: improve message when a program is deleted and is shared

Dittse, 2019-06-26:
If you try to delete or share one of your saved Programms, the corresponding window pops up for a very short amount of time and then disappears because the programm is loaded in the workspace.

[image]

boonto, 2019-11-25:
My first thought was to split it into two messages, one for your own programs. And one for programs shared with you.
However, as you can select multiple programs at once, including programs that are shared with others or shared with you, that would not make any sense.
I see three possible solutions:
1. One catch all message with all the info
2. 3 (your program, shared with you, multiple) messages with different info
3. Building up a fitting message depending on the selected programs

boonto, 2019-11-25:
I went with option 1.

bjost2s, 2019-12-07:
With this fix "POPUP_DISPLAY_HEADER" is declared a second time in blockly/robMsg/robMessages.js which leads to an error in the generated en.json. Please fix this.

boonto, 2019-12-08:
The duplicate message seems to have been introduced a year ago and adding the semicolon revealed it. I'll remove the duplication.

closed 2019-12-27 by bjost2s