OpenRoberta/openroberta-lab #130: Password reset bug: Email case sensitive

Dittse, 2019-06-26:
If the entered email address differs in upper and lower case from the one stored in the account, it will not be recognized.

[image]

[image]

rbudde, 2019-11-12:
It turns out, that according to stackoverflow "the part before the @ could be case-sensitive, since it is entirely under the control of the host system. In practice though, no widely used mail systems distinguish different addresses based on case. The part after the @ sign however is the domain and according to RFC 1035, section 3.1, name servers and resolvers must compare [domains] in a case-insensitive manner". Thus we can use a case insensitive compare.

But we have to guarantee, that no mail addresses are actually used in our database which differ only in upper/lower case.

kuruma99, 2020-03-16:
@bjost2s  I would like to work on this issue if it is still open.

boonto, 2020-03-16:
Hey @kuruma99 this issue is a pretty tough one and we are not really sure how to solve it yet ourselves, so it is definitely not suitable for beginners (although it seems easy).

rishabhdeepsingh, 2020-04-12:
@boonto  Maybe it will be great if we set new users' e-mail to lowercased and then at the front end whenever a new user resets the password that will be automatically lowercased and sent to the backend for reset.
For old users, we need to switch the case manually in the backend to the lowercase.

bjost2s, 2020-04-14:
@rishabhdeepsingh , as @rbudde already mentioned, we should not change the users email because the user might be unreachable or we might connect another user with the account.

rishabhdeepsingh, 2020-04-14:
That means we only need to check if there are some users with the same emails. Which I think is impossible as email providers don't allow this kind of thing.
Example: Example@gmail.com and example@gmail.com will be the same emails.

rishabhdeepsingh, 2020-04-14:
And if we have two different users with the same emails and different cases. The one with the original email can get the account of the Other if the other user sends a password reset request.
It is also possible to create two different accounts with the same email.
I created one with rishabhdeepsingh98@gmail.com and one with Rishabhdeepsingh98@gmail.com

rbudde, 2021-09-03:
you are right. See my comment from 12 Nov 2019.

We have the problem, that we HAVE users in the database with email, that differs w.r.t. upper/lower-case. Should we remove these? And the programs? Or move to one of the accounts? What if the name are the same, but programs differ?

rishabhdeepsingh, 2021-09-04:
I think we can move them to 1 account. or ask the user to move them in case they have access to all those accounts.
the naming of programs can be resolved by just replacing them with additional numbering at the end.

bjost2s, 2024-10-23:
because of compatibility issues it cannot be solved easily

closed 2024-10-23 by bjost2s