OpenRoberta/openroberta-lab #133: Scrolling toolbox not possible

Thorlogy, 2019-06-26:
With my asmartphone i can not scoll the toolbox.

[image]

Browser: Chrome and Firefox
OS: Android 8
Display: 6 Zoll (15,24 cm)
Resolution: 2160 x 1080 Pixel

philippmaurer, 2019-07-16:
I reproduced this bug in the iOS 10.3.3 Safari Browser and in the Roberta Mobile App on iOS and android.

bjost2s, 2019-12-26:
There is a fox available here https://github.com/google/blockly/issues/1174
Unfortunately this fix is based on a version of blockly, that we do not support, so applying it is impossible. 
Once we fully sync with latest blockly this problem will be solved automatically.

closed 2021-08-13 by bjost2s