OpenRoberta/openroberta-lab #135: make graphical code better readable by non-programmers by using correct grammar

Superwallah, 2019-06-26:
Some text blocks are not named in a way that they create a readable sentence when connecting those blocks.

1. sensor data objects (green with blue puzzle nose on their left side) like degree celsius of thermometer or brightness of light sensor should not start in the naming with "get" as they are built into the code as an grammar object, not as a relative clause including verb. 
Example: I want to check for the compass pointing North.
"If get angle° compass sensor = 0" is not understandable and does not seem "correctly programmed" for a kid. Why not can the block "degrees of compass"? Then it would be "If degrees of compass = 0" and "set k to degrees of compass" instead of "set k to get value° compass sensor".
I would advice to following changes:

- get xxx gesture ==> is moved xxx
- get angle° compass sensor ==> angle of compass (or "degrees of angle of compass")
- get sound% microphone ==> loudness from microphone (or "% of loudness from microphone")
- get value ms timer 1 ==> milliseconds of timer 1 (or "milliseconds of timer" as there is only 1 timer. And: kids don't know what "ms" is, "milliseconds" may give them a feeling.)
- get value° temperature sensor ==> degrees Celsius from thermometer
- get value signal strength ==> signal strength
- get value% light sensor ==> brightness  (or "% of brightness of light sensor")
- get analog pin 1 ==> analog pin 1
- get angle° gyroscope x ==> angle of x gyroscope (or "degrees of angle of x gyroscope")
- get value milli-g accelerometer x ==> acceleration in milli-g

2. It is possible to embedded blocks into others instead of appending at the end. So why not use it to allow the natural order of words.
I would advice to call it "wait xxx milliseconds" instead of "wait ms xxx".

3. "repeat 10 times do" are two verbs in one sentence. "repeat 10 times:" (colon at the end) will be clear enough, "do" as a word is not needed. Or use "repeat 10 times this:"
And: "repeat indefinitely this:" instead of "repeat indefinitely do".

4. "integer" is no term a newby understand. Why not call it "random whole number between xx and yyy" instead of "random integer from xx to yyy" ?

Shall I change it myself and offer a pull request? As there are 0 pull requests for this project, I am not sure if you would look at this. So I first wanted to discuss this here as an issue.

bjost2s, 2021-08-13:
we will rework the messages and create a more detailed issue for this

closed 2021-08-13 by bjost2s

Superwallah, 2021-08-13:
as a reference: can you please add a link of that new detailed issue to this old ticket?