OpenRoberta/openroberta-lab #137: copy and past of blocks BETWEEN different programs

rbudde, 2019-06-26:
I use a function to explain what I want. I have written a function and want to use it in another program.
Many solutions are possible, but at least one should be available:
- copy in one program, open another program, paste and get the function.
- save the function in the lab using name "myFunction" and have a new menu entry "import fragment"
- drag and drop between different browser instances.
I prefer the first solution.

bjost2s, 2019-07-02:
seems to be only related to #104

bjost2s, 2019-09-11:
item 1 (copy in one program, open another program, paste and get the function) is already available!
+1 for item 2 as an addition

corvus-albus, 2019-12-13:
Sorry for asking, but is it possible that item 1 is no longer possible? I knew that I once tried it successfully but either I do not remember how or it does not work any longer.
If it still works, it would be nice to hav some instructions in openrobertalab.
Cheers corvus-albus

closed 2021-08-13 by bjost2s