OpenRoberta/openroberta-lab #142: horizontal scaling with docker

rbudde, 2019-06-26:
in directory Docker there is a README describing how openroberta is put into docker container. There is also a description how to create these containers with docker, but that is not interesting here. The standard docker deployment is done with 2 container using docker-compose file dc-server-db-server.yml

- one container is the database server
- the other container contains the server and all crosscompiler

Calling the crosscompilers is likely the resource-intensive part. Thus the following idea arises:
- create a third container containing all the cross compiler (the dockerfile for that is already there, because it is used for creating the server container)
- deploy this container as often as needed

Now the "challenge": we need to add **load balancer support** to our (main) server. Standard load balancer won't work (pls convince me from the opposite ...). This is needed because we have to bundle 2 tcp-connections from different clients (robot + browser) and have to guarantee, that the program from the browser is sent to the crosscompiler container, that is used by the robot to retrieve the binary from. This is a non-standard kind of stickyness for a load balancer.

I propose first to assemble ideas, before we start any imlementation :-)

bjost2s, 2019-09-11:
related to #86

bjost2s, 2024-10-23:
no time to solve