OpenRoberta/openroberta-lab #144: Return compiler output as feedback to users

boonto, 2019-06-26:
For many robots the program is compiled serverside and we get errors/warnings from the compiler.
We should add functionality to display the compiler output to the user.
The different errors/warnings should be mapped to user friendly messages or may displayed raw by choice.

This would also remove the need of restricting user defined variable/function names in some cases (see 7 of #766): the user is notified of overwriting existing functions/variables/keywords.

rbudde, 2019-07-22:
beautification of the error message is needed, scrollbar or compiler specific cleanup. The latter must be an issue of its own for the next release.

Except that, resolved.

boonto, 2019-07-24:
Shows nice message that gives more information than previously: [image]

closed 2019-07-24 by boonto

ensonic, 2019-07-28:
Nice, but a few comments:
* can we make the dialog wider?
* can we replace the path of the tmpdir? Just having `/NEPOProg/... would be enough.

I also wonder that some people might be confused by the `[1/3]` output and expect 2 more messages.

bjost2s, 2019-07-28:
For one of the next releases we will consider a more readable output of the error messages. Because of many different compilers (and output of compilers) that we use, evaluating all compiler outputs in general is a big task, we need more time.