OpenRoberta/openroberta-lab #170: BOB3 compilation without arduino-builder

VinArt, 2019-06-27:
#34 - related

Currently bob3 uses arduino-builder to build and compile programs, though it is not necessary, since bob3 does not need a bootloader or other parts of Arduino framework. It works bare-metal.

Excluding Arduino framework will also make it easy to use the precompiled bob3 library, reducing the overall compilation to two compiler calls.

This can be achieved as follows:

1. Once:

```
avr-gcc -D_BOB3_ -std=c99 -g -Os -fno-exceptions -ffunction-sections -fdata-sections -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c analog.c -o analog.o
avr-gcc -D_BOB3_ -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c bob3.cpp -o bob3.o
avr-gcc -D_BOB3_ -std=c99 -g -Os -fno-exceptions -ffunction-sections -fdata-sections -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c ircom.c -o ircom.o
avr-gcc -D_BOB3_ -std=c99 -g -Os -fno-exceptions -ffunction-sections -fdata-sections -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c leds.c -o leds.o
avr-gcc -D_BOB3_ -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c main.cpp -o main.o
avr-gcc -D_BOB3_ -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c bob3tht.cpp -o bob3tht.o
avr-gcc -D_BOB3_ -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -DAVR -Wall -I.. -mmcu=atmega88 -DF_CPU=8000000 -c bob3smd.cpp -o bob3smd.o
avr-ar -q libbob3-smd.a analog.o bob3.o ircom.o leds.o main.o bob3smd.o
```

2. For each compilation:

```
avr-gcc -I../../../build/lib -D_BOB3_ -g -Os -fno-exceptions -fdata-sections -fno-threadsafe-statics -MMD -ffunction-sections -DAVR -Wall -Wextra -Wpointer-arith -Wcast-qual -Wmissing-include-dirs -Wno-unused-parameter -Wuninitialized -mmcu=atmega88 -DF_CPU=8000000 -finput-charset=UTF-8 -c teil3.cpp -o teil3.o
avr-gcc -L../../../build/lib/bob3 -mmcu=atmega88 -Os -Wl,--gc-sections -o teil3.elf teil3.o  -lbob3-smd
avr-objcopy -j .text -j .data -O ihex teil3.elf teil3.hex
```

courtesy of @springob

This, however would mean changes in code generation, since currently a proper Arduino sketch is generated and in this case a C/C++ program would be generated.

bjost2s, 2019-06-27:
where is the commit?
is it now released? We haven't tested it!

VinArt, 2019-06-27:
It is not resolved.

VinArt, 2019-06-27:
mixed up the projects, sorry

VinArt, 2019-07-02:
ArduinoSTL would not be compatible

VinArt, 2019-07-04:
So this is actually possible and compilation is instant. Lists cause the problem of applying this method.

VinArt, 2019-12-06:
- A small implementation of lists will be provided to substitute std::list
- Arduino builder will not be used

VinArt, 2019-12-11:
- there will be no lists for now

bjost2s, 2020-01-03:
works like a charm!

bjost2s, 2020-01-03:
Unfortunately math constants or more is not defined any more!
[image]

bjost2s, 2020-01-03:
In 1425720a9312df32aa90be2de912644fbcc6851d NEPODefs.h has been removed from code generation, this causes the above-mentioned error

closed 2020-01-03 by bjost2s