OpenRoberta/openroberta-lab #172: Since a couple of months EV3s (only EV3devs) cannot always connect to the server

bjost2s, 2019-06-27:
mainly during peaks, e.g. 10 to 12 in the morning. Seems that EV3Dev is trying to connect to the lab first with https and second with http. Our server has an proxy rule to redirect all incoming traffic to https unless it is a robot (identified by a specific string in the header) with old firmware (versions which haven't supported https on the robot). The last documented flag in the header was "... 2.7 ...", but it seems that EV3Dev is having "... 3.4 ...".

**To Reproduce**
Steps to reproduce the behavior: 
difficult, maybe change the code to connect only with html to reproduce the error

**Expected behavior**
 EV3Dev tries several times to connect and succeed finally.

This might work much more reliable if EV3dev tries to connect again with https instead of http. We could also change the rule in the apache for older versions.

ensonic, 2019-07-03:
A few details:
* only very old installations should use "ulrlib .. 2.7" as a user-agent string, newer ones should use "urllib 3.x". Ideally we don't hard-code the version to '3.4'
* the beahaviour switched from trying http first and then https to now try https first and only fallback to http
* I need to keep http support for developers testing to localhost and to users that use a raspberry pi
* CHanging the setting to include http or https would be nice but needs also ev3dev UI updates and that will take time to get released :/

The code in question is here:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/lab.py#L310

An alternative to trying the protocol could be to pattern-match the url and use https: for `*.openroberta.org` and http for anything else unless there already is a protocol prefix. This way people can put https://myserver.org into the UI and it will work.

Sounds like a plan?

rbudde, 2019-07-16:
In `/etc/apache2/sites-available` resp `sites-enabled` the file `lab.open-roberta.org.80.conf` was changed to filter a bit more general: `RewriteCond %{HTTP_USER_AGENT} !^Python-urllib/*`. The configuration was reloaded (thanx to Beate).

Thus, the problem should be fixed. @ensonic: pls inform us, if the problem remains.

Avoiding http in general would be a better solution. We should work on that in the future. If we disable ev3lejosv0 (which needs http always) somewhere down the line, this would be a good moment to close http access at all.

closed 2019-07-25 by bjost2s