OpenRoberta/openroberta-lab #177: German blocks in 'Robot Configuration' for SenseBox needed

vlebedynska, 2019-07-03:
Some actions blocks in 'Robot Configuration' for the SenseBox are only available in English. They should also be available in German.

bjost2s, 2019-07-03:
Where is the commit to blockly and what changes have been made in blockly_compressed.js?

vlebedynska, 2019-07-03:
> 
> 
> Where is the commit to blockly and what changes have been made in blockly_compressed.js?

Here is the commit link: https://github.com/OpenRoberta/blockly/commit/0051d1e36bad268e90c737156abb876799becbc8 and the pull request link: OpenRoberta/blockly#31.

@VinArt re-generated `blockly_compressed.js` because of changes made in the `de.json` file. No manual changes were made in `blockly_compressed.js`.

closed 2019-07-26 by bjost2s