OpenRoberta/openroberta-lab #179: Calli:Bot does not move backward when moved forward before

bjost2s, 2019-07-05:
**Describe the bug**
 Calli:Bot does not move backward when moved forward before. See also screenshot. Calli:Bot doesn't move after the first forward movement
**To Reproduce**
execute the program on a Calli:Bot and observe the behaviour

**Expected behavior**
 -Calli:Bot moves first forward and then backward
**Screenshots**
 - If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
I would like to do some more test concerning different movement sequences

bjost2s, 2019-07-11:
Seems to be a problem of the Calli:Bot design. 
Comment from the manufacturer: Please use accumulators instead of batteries, this will fix this problems.

rbudde, 2019-07-11:
Note: if the motors are stopped and then a short wait is executed, the program works for me.

It is questionable for me, whether we should support robots with such a weak design. Programming and debugging must be a nightmare for the user. Or?

bjost2s, 2019-08-02:
Following the the instructions from the manufacturer, Calli:Bot is working correct when using rechargeable battery instead of a normal battery.

closed 2019-08-02 by bjost2s