OpenRoberta/openroberta-lab #180: Revise and improve program checkers for all system

bjost2s, 2019-07-08:
A common solution for all systems is needed, which makes sure that checkers for all blocks with  parameters are available. This excludes special checkers for the program configuration relation which is handled in #114

We should start with EV3 to derive from this example the needed steps for all other systems.

boonto, 2019-11-08:
This issue changed in the process of the refactoring to lay the groundwork for revised and improved checkers. Now, additional checkers can be easily implemented as workers into existing workflows.

bjost2s, 2019-11-08:
Then this  issue is not resolved?

rbudde, 2019-11-11:
This issue is NOT resolved, but should probably be divided into some more concrete ones. It is too large and too unspecific.

rbudde, 2019-11-12:
overlap with #114

rbudde, 2019-11-12:
was tested. Remaining work put into new issue #306

closed 2019-11-12 by rbudde