OpenRoberta/openroberta-lab #182: Refactor ConfigurationComponent to contain blockly information

VinArt, 2019-07-11:
Currently robot configuration is handled in a different way to the program, it is not being transformed to an AST and back, so it is impossible to annotate the blocks, for example.

Information is only pulled from the XML and not written back.

In order for us to be able to annotate configuration blockly blocks, ConfigurationComponent needs to contain all the needed information to rebuild the blockly block.

VinArt, 2019-07-12:
see #114

VinArt, 2019-09-09:
ConfigurationComponent extends Phrase on feature/workflows

bjost2s, 2019-11-08:
There is a fix for this issue in the development branch, @VinArt please do not remove it from the project

bjost2s, 2019-11-08:
@VinArt , please provide some information how to test the solution

VinArt, 2019-11-11:
I added a unit test to check this.

closed 2019-11-12 by bjost2s