OpenRoberta/openroberta-lab #185: Show text with RGB values from colour sensor shows 0-100 range in simulation

VinArt, 2019-07-11:
**Describe the bug**
 - If one would print out the RGB values gotten from colour sensor on the EV3 screen in simulation the overall range is form 0 to 100 and not from 0 to 255

**To Reproduce**
- Create such a program:
[image]
- Run in simulation with robot view open

**Expected behavior**
 - Values would be from 0 to 255

**Device information**
 - OS: Linux
 - Browser Chrome 75

**Additional context**
Actual EV3 HAL implmentation has the following:
```
public synchronized ArrayList<Float> getColorSensorRgb(SensorPort sensorPort) {
        SampleProvider sampleProvider = this.deviceHandler.getProvider(sensorPort, ColorSensorMode.RGB.getValues()[0]);
        float[] sample = new float[sampleProvider.sampleSize()];
        sampleProvider.fetchSample(sample, 0);
        ArrayList<Float> result = new ArrayList<>();
        result.add((float) Math.round(sample[0] * 255.0f));
        result.add((float) Math.round(sample[1] * 255.0f));
        result.add((float) Math.round(sample[2] * 255.0f));
        return result;
    }
```
[image]
[image]

and since samples are from 0 to 1, then the overall result should be from 0 to 255

bjost2s, 2019-08-05:
see #74

rbudde, 2019-08-06:
see #215

bjost2s, 2019-09-23:
not solved in #215

bjost2s, 2019-10-02:
Please also check this in all robots systems, simulation and real robots!

boonto, 2019-12-17:
The simulation part was solved by a GCI student.

kuruma99, 2020-01-30:
If the issue is still open, I would like to work on this one.

boonto, 2020-01-30:
Do you have a real EV3 to test with? The simulation parts are already solved.

kuruma99, 2020-02-01:
No, I don't have one.

boonto, 2020-02-04:
Then you can't work on this sadly. Maybe check out another issue. I'll go through them soon and add some "good first issue" labels, so stay tuned for that.