OpenRoberta/openroberta-lab #188: Connecting leJOS 0.9.0 EV3 to the lab over WiFi/WLAN shows server error

philippmaurer, 2019-07-15:
**Describe the bug**
- Upon connecting an EV3 with leJOS 0.9.0 to the ORLab over WiFi, a "Server Error" message popup is shown.

**To Reproduce**
1. Go to the lab and select "EV3 leJOS 0.9.0" as robot
2. Connect the EV3 to a WiFi network (or WLAN) with access to the internet / ORLab
3. Connect the EV3 to the ORLab using the robot interface and enter the token in the ORLab
4. See the error message

**Expected behavior**
 - There should be no error message, since the connection to the robot is established (see *Additional context*)

**Device information**
- EV3 with leJOS 0.9.0
- WiFi/WLAN network with ORLab access

**Additional context**
 - Eventhough an error message is shown, the connection to the robot is still established. You just have to close the error message and press "cancel" in the EV3 connection form to get back to the programming interface. Afterwards you are able to run programs on the EV3.

bjost2s, 2019-07-15:
@boonto, I think we did this last Wednesday without problems?

philippmaurer, 2019-07-15:
@bjost2s I only tested this with one robot. Shall i test it with another one, too?

boonto, 2019-07-15:
We only tested EV3LejosV1, EV3Dev and C4EV3.

bjost2s, 2019-07-15:
It would be good to know, if we can reproduce this error

bjost2s, 2019-07-15:
@philippmaurer which version of EV3LejosV0 do you have?

philippmaurer, 2019-07-15:
Sorry for the inconvenience. It turned out that i got an experimental 0.9.0 version. (EV3 2.6.1-SNAPSHOT)
There is no error with the regular version.

closed 2019-07-15 by philippmaurer

gcielo, 2022-03-27:
I'm using leJos 0.9.1 and I can't find ORLab icon on the menu. How I can connect EV3 with OpenRoberta?