OpenRoberta/openroberta-lab #189: Can not transmit a program with "say" block to EV3 leJOS 0.9.0

philippmaurer, 2019-07-15:
**Describe the bug**
If the "say" block is used in a program and you try to transmit that program to an EV3 leJOS 0.9.0, an error message is shown and the program will not be transmitted.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the ORLab
2. Select EV3 leJOS 0.9.0 as robot type
3. Connect an EV3 leJOS 0.9.0 robot to the lab
4. Put the "say" block into the program
5. Press the "Play" btton to transmit the program to the EV3
6. See error

**Expected behavior**
 - The program should be transmitted to the robot and the EV3 should say whatever is specified.

**Screenshots**
[image]

**Device information**
 - EV3 leJOS 0.9.0, connected over WiFi/WLAN

**Additional context**
 - A program with the "say" block does work in the simulation, but the "say" has no effect, as if it was skipped. THis is the same for EV3 0.9.1

bjost2s, 2019-07-15:
The say block is only implemented for EV3 0.9.1 and EV3Dev and it definitely cannot be implemented in EV3 0.9.0, so we should improve the error message for this case!
The simulation supports all blocks from EV3 0.9.0 with the standard configuration, so far
and also I think there was an issue with text to speach in the browsers, we will check it again.

bjost2s, 2019-07-18:
unfortunately this block has no help :(, I think that will be also a good place to explain the situation to the user

boonto, 2019-08-01:
Usually there should be a warning annotation for EV3LejosV0, which notifies the user but allows compilation and running the program. But that is missing.

The missing simulation speech is handled in #204

boonto, 2019-08-01:
After fixing the old approach the message says `The exection of this block will have no effect!`, but `Überprüfe die Einstellungen bei diesem Block. Er wird so nichts ausführen können.` in german. The english message sounds fine, however the german one does not. @bjost2s should we introduce a new message `This robot variant does not support this block. It wont be executed.`?

closed 2019-09-06 by bjost2s