OpenRoberta/openroberta-lab #191: When using the "play frequency Hz" block on WeDo, the program will not be executed

philippmaurer, 2019-07-16:
**Describe the bug**
If a program contains the "play frequency Hz" block on WeDo, the whole program does not run on the WeDo and it is no longer possible to start another program on it. The robot needs to be reconnected.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the Open Roberta Mobile App
2.  Connect a WeDo to the App
3. Create a valid program and add the "play frequency Hz" with a valid input to the program
4. Press the play button to run the program on the WeDo
5. The WeDo will do nothing.
6. The Play button will be grayed out.
7. To run programs on the WeDo again you need to reconnect it.

**Expected behavior**
 - The program should be executed, including the "play frequency Hz" Block.
 - It should be possible to start other programs on the WeDo after the program ended.

**Screenshots**
[image]

**Device information**
 - WeDo 2.0
 - Open Roberta Mobile App
 - iOS & Android

bjost2s, 2019-08-05:
see #74

rbudde, 2019-08-07:
See #215

closed 2019-08-07 by rbudde