OpenRoberta/openroberta-lab #192: The Roberta Mobile App freezes, if a loop is used without "wait ms" blocks in it

philippmaurer, 2019-07-16:
**Describe the bug**
If a program with a loop in it is executed on the WeDo, in which the loop does not contain a "wait ms" block, the Roberta Mobile App freezes during the loops execution and the WeDo does not no what is specified in the loop.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the Open Roberta Mobile App (App)
2.  Connect a WeDo to the App
3. Create a valid program that contains a "repeat until" loop, or another loop that runs for a long time. There must not be a "wait ms" block inside the loop.
4. Press the play button to run the program on the WeDo.
5. The code before the loop *may* be executed fine, if it is fully executed before the loop starts.
6. After that, the App will freeze and the WeDo will not get new commands.
a. If the "repeat until" loop was used, the WeDo does not react to the break condition anymore.
b. If a loop with a static break condition was used (for loop with 10k iterations for example), the app will continue fine after the loop is fully executed. However, whatever is specified in the loop that would run on the Wedo, will not be executed.

**Expected behavior**
 - The loop should be executed normally and the App should be usable during its execution.
 - The WeDo should be able to read its sensors during the execution.

**Screenshots**
[image]

(Here, the WeDo wil do nothing. The text will never be shown. The App needs to be restarted.)

**Device information**
 - WeDo 2.0
 - Roberta Mobile App
 - iOS & android

bjost2s, 2019-08-05:
see #74

rbudde, 2019-08-07:
see#215

closed 2019-08-07 by rbudde