OpenRoberta/openroberta-lab #197: Create list with item int repeated N times doesn't compile on Arduino

simonedegiacomi, 2019-07-17:
**Describe the bug**
The block to create a list with an item repeated N times doesn't compile on Arduino if the item is an int.
This is because the utility function `_createListRepeat` get expanded to return a `std::list<int>`, which cannot be casted to `std::list<double>`.

**To Reproduce**
1. Create a new program and select one of the Arduino targets
2. Connect an Arduino
3. Create the following program
[image]
4. Press Run and see the compilation error

Notice that the following program compiles:
[image]

**Expected behavior**
The program should compile.

**Screenshots**

[image]

**Device information**
 - Type: Desktop
 - OS: Ubuntu
 - Browser: Firefox

**Additional context**
Bug discovered while working on #29

simonedegiacomi, 2019-07-17:
I resolved the same problem in c4ev3 with this commit https://github.com/OpenRoberta/openroberta-lab/commit/b56181d76bb4bc4fb28f9cbbc1b30e44063db032. If it is okay I can move this code into AbstractCppVisitor to fix the block also for Arduino

bjost2s, 2019-07-17:
Sounds good. Please check if (float) would be ok, too. I think we use float in general on the robots unless it is defined differently. Please also check the IT tests, they should have covered this.

rbudde, 2019-07-18:
Nope. Actually nosystematic test coverage for lists. A student ist working exactly on this topic, if the other tests he is working on are complete (hopefully in the near future)

simonedegiacomi, 2019-07-23:
I updated the title, since the program doesn't compile also with string. A cast is necessary, otherwise the following error appears:
```
error: conversion from ‘list<const char*,allocator<const char*>>’ to non-scalar type ‘list<std::__cxx11::basic_string<char>,allocator<std::__cxx11::basic_string<char>>>’ requested
```

simonedegiacomi, 2019-07-23:
@bjost2s I checked about double or float and it appear that in Arduino double it's used for numbers and list of numbers

VinArt, 2019-09-03:
This is compiling (at least) now and is already in develop

closed 2019-09-04 by VinArt