OpenRoberta/openroberta-lab #199: Access to all of pins on the mirco:bit board

munka21, 2019-07-18:
**Is your feature request related to a problem? Please describe.**

When micro:bit will be connected with a IO F5 board and if you want to program with this board, you need to have a access to all of pins, which are on the micro:bit available. It is necessary if you want to program a different model, such as the model from fischertechik. But in NEPO you have access to only 3 of 16 pins available. 

[image]

[image]

**Describe the solution you'd like**
 I would like to have a access to all of pins on the micro:bit board

bjost2s, 2019-07-18:
Interesting, so far we do not have an appropriate board to test this, so the implementation will be hard. Once we can get the hardware we will come back to you request

boonto, 2019-08-20:
I followed [these](https://microbit.org/guide/hardware/pins/).

VinArt, 2019-09-04:
I think, before resolving this, we should test how it works. I am not quite sure, whether it is possible, for example to use pins 19 and 20 for GPIO, since they are connected to I2C.

closed 2019-09-16 by VinArt