OpenRoberta/openroberta-lab #203: Simulation does not open if ternary block is in task

boonto, 2019-07-24:
**Describe the bug**
 - If a ternary block is in task the simulation does not open.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a ternary block to any block in the program
2. Try to open the simulation

**Expected behavior**
 - The simulation opens.

bjost2s, 2019-08-05:
here are two programs where the problem does not show up:
EV3
[image]
Calliope
[image]

Please provide an example

boonto, 2019-08-05:
I just checked, one of the branches has to be empty.
Works:
[image]
Does not work:
[image]

rbudde, 2019-08-07:
See #215

VinArt, 2019-09-23:
Simulation does not open, ternary block has wrong colour of output.

bjost2s, 2019-10-02:
Will be solved with EmptyBlockVisitor

closed 2019-10-02 by bjost2s