OpenRoberta/openroberta-lab #211: Open Roberta Lab does not "load" in Internet Explorer

philippmaurer, 2019-08-01:
**Describe the bug**
When opening the Open Roberta Lab on the Internet Explorer (testet IE Version 11), the loading indicator at the beginning never stops turning.

**To Reproduce**
1. Open the Internet Explorer
2. Go to the Open Roberta Lab
3. See error

**Expected behavior**
 - The Lab should load and be usable.

**Device information**
 - Internet Explorer 11 (Tested V10 via the Compability Mode of IE as well, same error)
 - Windows 10

**Additional context**
 - The multiSimController.js uses "const" instead of "var", which is not supported in IE 10. Therefore the page will not be usable on IE 10 as long as this feature is used. Replace it with "var" or use a polyfill. 
   - This should not be the problem for it not running in IE 11 though.

bjost2s, 2019-08-01:
This is a problem of not supported ECMAScript 6 in IE and IOS Webviews. Will be fixed asap

rbudde, 2019-08-01:
startWith is not available in IE<12; wrong behavior in some IOS webview impl.
Replaced by indexOf

Added to 3.5.1 for a later review

closed 2019-09-04 by VinArt