OpenRoberta/openroberta-lab #222: document url params

ensonic, 2019-08-15:
**Describe the bug**
The implementaion in [menu.controller.js](https://github.com/OpenRoberta/openroberta-lab/blob/bf7ca882137cff3a065e75a28fce833578841e1c/OpenRobertaServer/staticResources/js/app/roberta/controller/menu.controller.js#L25) supports a few parameters that can be passed to the lab.open-roberta.org url. These are useful for codingHubs to configure computers for their courses.

**Expected behavior**
Right now we don't document them neither in the wiki nor in the actual documentation, but we should.

**Additional context**
Before we document them, do we want to reconsider the format a bit?
See this comment
https://github.com/OpenRoberta/robertalab/issues/924#issuecomment-399214219

rbudde, 2019-08-15:
_I support your claim to reconsider the format_

boonto, 2020-01-07:
After some groundwork from a GCI student (thanks @KevinL10) I added this wiki page: [URL parameters](https://github.com/OpenRoberta/openroberta-lab/wiki/URL-parameters).

closed 2020-01-07 by boonto